p-v-o-s/sky-sweep #2: Create BOM

dwblair, 2015-06-05:


Fastie, 2015-06-14:
Don,
I updated the SkyShield [bill of materials google sheet](https://docs.google.com/spreadsheets/d/1GBuHSpaCLA2EbN3OOJWe4kTiJWiZ1Trc5mMFSvcZXjI/edit#gid=2020475328). There are now sheets for Version 0.4 (in beta), Version 2.0 and 2.1 (at OSH, will have DIP switch) and Version 3 (not designed, will have boost converter). It would be good to place a Digi-Key order for DIP switches, resistor networks, and other stuff for Version 2 in the next few days. Could you source the resistor network, confirm my DIP switch choice, and confirm that Version 2 still needs the resistors and capacitors that version 0.4 needed?

I could also include in this Digi-Key order a boost converter and anything else needed for Version 3. That is not designed yet, but I hope it will be in time to order some PCBs from OSH as soon as Version 2.1 is confirmed to work (this month). That way we could have Version 3 back from OSH and ready to try when I return from field work. In fact, if you come up with a design for Version 3 like now, I will order a few and maybe get to test it before I ship out, which could move the timeline up by two weeks.

There are notes to you in the Google sheet.
Thanks

dwblair, 2015-06-14:
Great!  I'll get on resistor network, DIP switch, and associated resistors and capacitors ...

Love the boost converter idea.  I think the part that was pointed to was a 'surface mount' part, and I haven't looked at the associated 'package', but depending, it might be more or less easy as a DIY kit.  @bgamari  has pointed to the fact that some surface mount packages are actually easier to solder than through-hole, though, so maybe this will work out well ... will look into it.

Fastie, 2015-06-14:
Here are some [through-hole step-up boost converters at Digi-Key](http://www.digikey.com/product-search/compare/en?returnUrl=%2Fproduct-search%2Fen%3FFV%3Dfff40027%252Cfff8029a%252C1140050%26mnonly%3D0%26newproducts%3D0%26ColumnSort%3D1000011%26page%3D1%26stock%3D0%26pbfree%3D0%26rohs%3D0%26quantity%3D%26ptm%3D0%26fid%3D0%26pageSize%3D25&part=497-4280-5-ND&part=296-36112-5-ND&part=296-17766-5-ND&part=497-7414-5-ND&part=296-17764-5-ND&part=497-7412-5-ND&part=497-7413-5-ND&part=MC34063AP1GOS-ND&part=KA34063AFS-ND&part=NJM2374AD-ND&part=TL7660CPE4-ND). These are all 8 pin ICs (7.6mm), so they take up a lot of room on the shield.   But they all cost <$0.65.

Fastie, 2015-06-14:
[Here are more](http://www.digikey.com/product-search/compare/en?returnUrl=%2Fproduct-search%2Fen%3FFV%3Dfff40027%252Cfff8029a%252C1140050%26mnonly%3D0%26newproducts%3D0%26ColumnSort%3D1000011%26page%3D2%26stock%3D0%26pbfree%3D0%26rohs%3D0%26quantity%3D%26ptm%3D0%26fid%3D0%26pageSize%3D25&part=MC33063AP1GOS-ND&part=TC7660CPA-ND&part=TC7660EPA-ND&part=LMC7660IN%2FNOPB-ND&part=TC7660SEPA-ND&part=TC7660SCPA-ND&part=TC1044SCPA-ND&part=TC1044SEPA-ND) for mostly <$1.00 includiing some that are ratiometric, whatever that means. These are all through-hole ICs. Could these be soldered on the underside of the shield to allow something to go in its footprint on top?

bgamari, 2015-06-15:
@Fastie, be aware that the second group of parts (the "ratiometric" ones) actually produce a negative voltage (this type of device is known as a switched capacitor charge pump). You really want something like the parts in the first list, which are [buck-boost regulators](https://en.wikipedia.org/wiki/Boost_converter) (where you are instead using an inductor as your energy storage element).

Fastie, 2015-06-15:
Thanks Ben. Good to know what ratiometric means, and that we don't want it.

bgamari, 2015-06-15:
Well, to be clear it's not that you don't want a ratiometric part. "Ratiometric" in this case just means that the output voltage is proportional to the ratio of two external component values (in this case resistors).

Also, it seems I mixed up the two sets of parts that you cited. You can filter for the desired parts on Digikey's website by selecting the  "Boost" and "Buck-boost" topologies.

Fastie, 2015-06-15:
Hmm, all the boost and buck-boost through-hole converters are expensive ([cheapest ones](http://www.digikey.com/product-search/compare/en?returnUrl=%2Fproduct-search%2Fen%3FFV%3Dfff40027%252Cfff8029a%252C1140050%252C11280136%252C11280138%26mnonly%3D0%26newproducts%3D0%26ColumnSort%3D1000011%26page%3D1%26stock%3D1%26pbfree%3D0%26rohs%3D0%26quantity%3D%26ptm%3D0%26fid%3D0%26pageSize%3D25&part=296-36042-5-ND&part=296-3226-5-ND&part=LT1303CN8%23PBF-ND&part=LT1303CN8-5%23PBF-ND&part=LT1301CN8%23PBF-ND)).
The surface mount versions are way more abundant and cheaper ([cheapest ones](http://www.digikey.com/product-search/compare/en?returnUrl=%2Fproduct-search%2Fen%3FFV%3Dfff40027%252Cfff8029a%252C1140003%252C11280136%252C11280138%26mnonly%3D0%26newproducts%3D0%26ColumnSort%3D1000011%26page%3D1%26stock%3D1%26pbfree%3D0%26rohs%3D0%26quantity%3D%26ptm%3D0%26fid%3D0%26pageSize%3D25&part=NCP1402SN33T1GOSTR-ND&part=NCP1400ASN30T1GOSTR-ND&part=NCP1400ASN27T1GOSTR-ND&part=AP3015AKTR-G1DITR-ND&part=AP6714M10GDITR-ND&part=863-1496-2-ND)).
Can we ask kit builders to solder on a surface mount component?

BobStevens, 2015-06-15:
I've been using the mini buck-boost modules from Pololu, on an unrelated
project, and they seem to work well. Easy to solder.
When used with servos an additional capacitor on the Vout helps a lot. But
perhaps too expensive to include in the kit.

The board and schematic can be found here...
https://www.pololu.com/product/2119/resources

bgamari, 2015-06-15:
Looks like they are a bit expensive and may not have the capacity to drive the load in question.

Fastie, 2015-06-16:
Thanks for that link Bob. That Pololu module seems to do just what we need, although it would be good to be able to handle about 1.5A. I haven't been able to find one of the IC modules at Digi-Key that is step-up/step-down and in the correct voltage range for a reasonable price. The servos seem to need between 4.8 and 6 volts. I think closer to 6V is better. There are lots of IC converters for under $2.00, but I'm not sure any of them do what we need. I'm still not sure adding $5.00 to the BOM is justified. I also don't know which of those surface mount converters can be soldered by hand.

I just got the news that six new SkyShield PCBs were shipped from Oregon today. So the more pressing need is the final components for SkyShield v2 and 2.1 (DIP switch, resistor network, etc.). A Digi-Key order should go in ASAP, then we have more time to figure out if a boost converter can be added for v3.

dwblair, 2015-06-16:
​Hallo!

Yes, let me get back in here and review the components for v2 and v2.1 ... ​

dwblair, 2015-06-16:
Good news: confirmed number of capacitors, added in resistor network option
on digikey!

Potentially bad news: in discussion w/ Ben, might've done something that
won't quite work w/respect to the DIP switch. Going to prototype it tonight
to find out ...

dwblair, 2015-06-17:
Update:

When I checked the parts for v.2.1 and v.2.2, something looked odd to me in
the switch circuit.  Specifically, there was no "ground" for any of the
switches -- so that when the switch is open, the voltage measured on the
analog pins will 'float'.  (This is in contrast with our other switch /
button circuit, which is grounded when it's not closed -- and is what
tipped me off.)

In conversation with @bgamari (The Oracle), he proposed a simple fix for
this.  Arduino pins are capable of 'internal pull-up resistors' -- i.e.,
through firmware code you can decide to connect individual pins to VCC via
an internal resistor.

So: if we connect the individual switches in the DIP switch to ground,
instead of VCC, we can also 'pull up' all of the resistors via an internal
pull-up resistor in firmware.

This doesn't work as a fix for boards version 2.0 and 2.1 -- in order to
fix those boards, you'll need to 'cut the trace' (the PCB metal connecting
the switches to VCC) on the board (done with a knife or equivalent), and
then solder a wire from the relevant pin on the resistor network to GND.
This won't be a great option for a commercial kit, and you likely won't
want to sell boards with this fix (though I'm sure you can do it in an
elegant way) -- but @bgamari pointed out that it might be worth ordering
the resistor network part in order to test the circuit at least once,
anyway.

The upside of this is that we can actually get away without using the
resistor network at all, making more space on the board (for, e.g., the
components required for a switching regulator -- which will require not
just the switching reg IC, but some additional capacitors / inductor / etc.)

Sorry for the goof-up.

dwblair, 2015-06-17:
Here's a new version of the shield (2.2) incorporating the changes
(simplified switch circuit):

https://github.com/p-v-o-s/sky-sweep/tree/master/ver_2.2/hardware

bgamari, 2015-06-17:
dwblair <notifications@github.com> writes:

> Here's a new version of the shield (2.2) incorporating the changes
> (simplified switch circuit):
>
> https://github.com/p-v-o-s/sky-sweep/tree/master/ver_2.2/hardware
>
Looks good to me.

Fastie, 2015-06-17:
So there is code you use in a sketch to make a particular pin gets its current through a resistor on the Nano? Only the pins used for the DIP switch will be routed through this resistor? Anybody know this code? 
With some scratches and jumpers can I test this with the PCBs that will arrive soon? That is, without a resistor network, can I wire it up so the DIP switch works? Are any other components needed for this test? 
If the next version will not have a resistor network, is there any reason to order one for testing?

BobStevens, 2015-06-17:
INPUT_PULLUP is the mode,
http://www.arduino.cc/en/Reference/PinMode

dwblair, 2015-06-17:
​Actually, great point -- there's no need to order the resistor network. In
fact, *not* placing the resistor network makes ​testing easier -- you can
connect a wire from one of the resistor network holes directly to ground,
without having to cut any traces.  So really, you only need wires and
solder.  There is one pin on the resistor network footprint that is
connected to VCC, but each of the other pins would simply be connected to
ground (each separately).

Thanks Bob for the link -- yes, that looks like the right code, there!

Fastie, 2015-06-17:
So for a 3 switch DIP, I will need 3 jumpers, each from ground to a different hole (where the resistor network would have been). Wouldn't that be the same as connecting all the holes and running one jumper to ground? Could you doodle the jumpers required? 

One jumper should not bother potential beta testers, although the plan was to have this round of testers get a kit for soldering the PCB. This version might not make a good test of that.

 I will place an order for DIP switches, but I have enough of everything else. 

I guess I could order some version 2.2 PCBs from OSHPark, but I won't get to try them until late July. Will there be a version 3 with boost converter by then? Or is finding a boost converter for <$5.00 going to be hard?

dwblair, 2015-06-17:
Ah, you're right -- all of the switch ends can be connected to ground together, so that's easier. 

For some reason I'm unable to change the line color in Inkscape any longer. Oops.  But hopefully you can discern what I'm getting at here:

[image]

[image]

Not yet sure re: the boost converter ... but sounds reasonable to shoot for late July, fi we can find a cheap part ...

Fastie, 2015-06-17:
Thanks for the doodle. The jumpers could be underneath so they won't even show. For v 2.1 with a four-switch DIP, it should be as you show.  For v. 2.0 with a three-switch DIP, can I connect only three holes (which three)? 
It sounds like this could be a quite functional SkyShield and the DIP switch is an important upgrade to test. I have all the parts for six new rigs printed, so I could run another beta test. I probably won't have time to assemble the rigs before I ship out, but maybe beta testers can test the whole kit process this time. I just have to create assembly instructions and the DIP switch code. Bob, thanks for the link to the  PinMode page.