p-v-o-s/sky-sweep #3: USB power flakiness -- exploring

dwblair, 2015-06-05:


Fastie, 2015-06-07:
I added to the description of this power issue sufficient detail that someone familiar with circuits and power could probably diagnose it.

bgamari, 2015-06-09:
Indeed most USB ports are limited to 500mA. A standard servo may draw upwards of an amp when starting. This almost certainly explains the behavior you see.

Also, something to note is that even when not moving some servos can draw quite a bit (tens of milliamps). A simple solution to both of these problems would be to put the servos behind a switch-mode boost converter. This will allow you to use fewer batteries (as the battery voltage is no longer constrained by the voltage requirements of the servos) and allows you to easily switch off the servos when not needed.

Fastie, 2015-06-09:
Thanks Ben. So USB 2.0 just can't supply the current that servos need. 

A boost converter would solve that problem and supply more reliable power from batteries as well. It looks like a boost converter would cost as much as all the components of the SkyShield combined and be almost as big as the entire SkyShield. 

I guess since the SkyShield functions as desired under battery power, which is the intended mode for aerial photography, it might be best to leave it as is. Testing new sketches downloaded via USB will require that battery power is on (whether USB is still connected or not).

Fastie, 2015-06-09:
Maybe we shouldn't lay all the blame on USB. USB connects to the Nano, not the SkyShield, so USB power getting to the pins on the shield goes through the Nano which is rated for only 40mA output. That is certainly not enough to drive servos, and a primary function of the SkyShield is to supply adequate power to the servos because an Arduino alone can't do that. 

The 1.5A LDO voltage regulator on the SkyShield is a better power source than USB, but the 500mA direct from USB should be enough to handle the servo motion it is being asked to do. 

Unless there are output pins on the Nano that could supply the USB current directly to the SkyShield, there are no changes to the SkyShield that could fix this problem. The servos will never work well if all the power they get is from the 40mA Nano output. If there is a pin on the Nano that supplies unfiltered USB power, the SkyShield could direct that to the servos, with or without regulation or boosting. Is there such a pin?

There must be, because the Funduino shield drives servos just fine and has headers intended to do just that. With USB connected to the Nano, servos plugged in to the Funduino operate as they should. What does the Funduino do that the SkyShield doesn't?

Fastie, 2015-06-10:
From this post: <http://electronics.stackexchange.com/questions/67092/how-much-current-can-i-draw-from-the-arduinos-pins>

"The 5V pin of the arduino is *not connected through the microcontroller.* As such, it can source significantly more power. When you are powering your arduino from USB, the USB interface limits your total power consumption to 500 mA. This is shared with the devices on the arduino board, so the available power will be somewhat less...

...The 5V output pin is good for ~400 mA on USB, ~900 mA when using an external power adapter."

I think the SkyShield uses this 5V pin on the Nano to get power when it is connected to USB. So it should have about 400mA to work with. That's not overly generous for servos, but should be plenty for testing. Is it possible that the SkyShield is preventing some of this current from getting to the servos?

bgamari, 2015-06-10:
Hmm. A simple boost converter should cost around a dollar with all of the passives. The converter itself will be roughly 20mm^2, with perhaps another 20mm^2 of routing and passives. Surely the board isn't *that* small. Moreover, the potential reduction in battery size would likely outweight the larger board area.

As far as the servos go, the problem may be a transient high current requirement. It is possible that some capacitance on the 5V rail may smooth out this transient

bgamari, 2015-06-10:
For instance, this would be one possible switcher, http://www.ti.com/product/lmr61428.

Fastie, 2015-06-10:
Ah, 60¢ and 3x5mm. That's more like it. My searches were finding something else.  That's even cheaper than the LDO on the SkyShield now. 

The LDO is in the circuit right after the battery input, but that has no effect on power coming from USB. The boost converter could replace the LDO but get behind both the battery and the 5V pin so any power going to the servos gets boosted.  

The battery we're using now is minimal because 4 AAA alkaline are only 6V and the servos should have at least 4.8V.  So when the batteries get a little worn out they stop driving the servos properly. I guess a boost converter could allow those same AAA to last much longer. 

It sounds like a worthwhile addition.

bgamari, 2015-06-11:
Indeed, you could drop that down to 2 batteries at the expense of some capacity, some of which you would win back as the boost converter can maintain adequate rail voltage even as the battery voltages sag. With some effort you could probably find a boost converter with a minimum voltage of 1V, allowing you to operating with one alkaline battery although it's not clear it would give you enough capacity.

Fastie, 2015-06-11:
It might be best to stay with the four AAA so users can be confident that they have enough power to last for an afternoon of flying. With just two batteries, you would probably want fresh batteries for every one hour flight. Eventually you will send the rig up for an hour and then learn that the batteries lasted only 10 minutes, but with four AAAs and the right boost converter, that will happen on fewer flights.

Boost converters are available in through hole configuration, but there are zillions of them at Digi-Key. Our current LDO is mounted underneath the SkyShield so it takes up little space (it ends up between the SkyShield and the Nano).   
  
[image]  
   
Don, can you find an appropriate converter for 4 AAA and figure out if can fit on the SkyShield? The new SkyShield v2.1 will arrive in the next couple of weeks so its time to put in a Digi-Key order for components for those. So we need to select a resistor network and DIP switch, too. I can include in that order a couple of boost converters to try out for SkyShield V. 3.0.

dwblair, 2015-06-18:
Specs on this particular chip look maybe okay? https://learn.adafruit.com/adafruit-powerboost-1000-basic

Fastie, 2015-06-18:
It would be slick to be able to use two AAA instead of four. It would also be nice to use rechargeable batteries (which are 1.2V instead of 1.5V for alkaline). But that module is about half the size of the entire SkyShield, and costs more than all the other components combined (including the Nano).

There are two reasons to change the power circuit.

 -   1) fix the issue about USB power not driving servos correctly, and 
 -   2) supply more carefully regulated power so batteries last longer and so fewer batteries are needed.

Is it possible to move the LDO we now use so that it regulates the power coming from the USB port on the Nano? Would this fix problem 1?

The existing configuration works really well in flight. Four AAA batteries last for at least 1.5 hours and maybe much longer (not yet tested). Lifting four AAA is not an obstacle to kite photography most of the time. So that part (number 2) ain't really broke. 

I don't think it's worth adding 50% or 100% to the BOM to fix either 1 or 2, but if there is a way to fix problem 1 without adding more than a few dollars and enlarging the SkyShield footprint, it would be great to find it.

bgamari, 2015-06-18:
@Fastie I suspect Don was suggesting the chip on that board (the [TPS61030](http://www.ti.com/product/tps61030)) and not the board itself. It is [available](https://octopart.com/search?q=TPS61030) in a TSSOP package in single quantities for about $3. It's certainly not the cheapest option but I agree that the specs look pretty nice.

Fastie, 2015-06-23:
Just came across [this on Adafruit](https://learn.adafruit.com/adafruit-arduino-lesson-14-servo-motors/if-the-servo-misbehaves):
"Your servo may behave erratically, and you may find that this only happens when the Arduino is plugged into certain USB ports. This is because the servo draws quite a lot of power, especially as the motor is starting up, and this sudden high demand can be enough to drop the voltage on the Arduino board, so that it resets itself.
If this happens, then you can usually cure it by adding a high value capacitor (470uF or greater) between GND and 5V..."

Ben suggested this above 13 days ago. Is there any sacrifice to adding a big capacitor like this? 
[These cost <50¢ ](http://www.digikey.com/product-search/compare/en?returnUrl=%2Fproduct-search%2Fen%3FFV%3Dfff40002%252Cfff80009%252C3400b7%252C1140050%26mnonly%3D0%26newproducts%3D0%26ColumnSort%3D1000008%26page%3D1%26stock%3D1%26pbfree%3D0%26rohs%3D0%26quantity%3D%26ptm%3D0%26fid%3D0%26pageSize%3D25&part=P5155-ND&part=P5168-ND&part=P10221-ND&part=P12376-ND)

bgamari, 2015-06-23:
Fastie <notifications@github.com> writes:

> Just came across [this on Adafruit](https://learn.adafruit.com/adafruit-arduino-lesson-14-servo-motors/if-the-servo-misbehaves):
> "Your servo may behave erratically, and you may find that this only [...]
There isn't much of a sacrifice beyond the volume of such a capacitor,
per se. That being said, I would argue that a switching regulator offers
far more in terms of benefits. Not only is it substantially more
efficient than an LDO, but it allows you to use fewer batteries for
a larger portion of their discharge curve.

A switcher may not be quite as simple as a simple cap but you can be fairly certain that
it will work (in contrast to the cap) and it offers a number of
additional advantages.

Fastie, 2015-06-23:
Yes, those are big caps, typically 8mm diameter and 16mm tall.  If placed cleverly, one could fit between the Nano and the shield and add nothing to the footprint. And it would add only 50¢ to the BOM, keeping it under $10.00 total. 

I have not yet seen a description of how many components would be required to implement a boost converter, how much space they would require, and how much they would cost. 

I'm not capable of that circuit design, so i think the SkyShield with a boost converter is a second generation device. If Don can incorporate the deck chair changes we outlined in [another thread](https://github.com/p-v-o-s/sky-sweep/issues/12), I can order a few v2.3 PCBs this week so they are available when my field work is done mid July. If they test well, I might order a bunch and offer the first retail kits for sale in early August before the KAP season ends. 

After v2.3, the next priority could be to breadboard a big capacitor to see if it helps. Don has all the components he needs for that experiment except for the 470 uF cap. 

In the meantime, version 2.0 is in beta for anybody who wants one: <http://publiclab.org/notes/cfastie/06-23-2015/skyshield-v2-beta>