p-v-o-s/sky-sweep #9: Camera USB cable attachment to SkyShield header is lame.

Fastie, 2015-06-06:


dwblair, 2015-06-06:
Well I never.

(ooh -- should we use a microUSB thingy?)

Fastie, 2015-06-06:
Pat faced one of the more tricky aspects of operating the Saturn V Rig on his first attempt to fly it.  The standard way to connect a cable to a PCB is with a header. One or a few pins stick up from the header and a plastic connector fits over the pins. The only thing that holds the two together is metal-to-metal friction. The camera cable is stiffer than typical servo leads, and when the camera tilts it pulls on the cable. The connector needs only two pins, so it is not as sturdy or tight as a three pin connector.    To compensate for this, the cable has a Velcro strap on it which can hold the cable firmly in place.  

During launch, you might need to press the button on the SkyShield to select the mode, and that requires moving the SkyShield which can dislodge the camera cable connector. Trying to plug that connector in again while the kite is flying and the Picavet lines are getting wrapped around the turning gears is tricky and can result in the header pins getting bent. It's easy to bend the pins straight again, but eventually they will divorce themselves from the PCB.

The interim solution is to avoid last minute access to the SkyShield. Get familiar with plugging that connector in and get everything set before the kite flies. Of course selecting any mode other than Mode 0 will always require access to the button (but that will change with SkyShield v2.1).

Is anyone familiar with better connectors that could replace the standard servo connectors and be really cheap and easy to use (kit builders will have to make their own camera cables)?

Fastie, 2015-06-08:
Don, a micro USB connector could work instead of the present header. I can buy cables ready made for that (male to male, right angle USB mini to straight USB micro, 25 cm long).  But those cables cost $2 more than the ones I have been modifying, and the USB micro connector costs a dollar more than a header. Plus it's hard to mount a USB connector vertically and make it strong.

dwblair, 2015-06-08:
​hmmm ... could we shift the switch and place two mini usb headers facing
'out' at the 'bottom' of the board?

b/c these are pretty cheap
http://www.monoprice.com/Product?p_id=5451&catargetid=320013720000066114&cadevice=c&kpid=105451&gclid=CO2m5OLv_sUCFZWPHwod1Y8ApQ

but you're right, the usb connectors do cost more ...

Fastie, 2015-06-08:
I think the cable we need is right angle mini B (for the camera) to micro B (for the shield), and about 25 to 30 cm long. We could use mini B instead of micro B on the shield but I assume those are bigger.

However, USB on the shield is really more than we need. We need to connect only two wires, and space and accessibility are limited (USB connectors are fat). A simple snap on connector should be easy to find for $0.30 and then we have to solder its mate onto the other end of a right angle mini B USB cable. 

Although Pat is having repeat trouble keeping that cable connected during flight, I think it will work for him eventually.  So unless there is consensus that the current cable is unworkable, we would have to find a really nice substitute for the ubiquitous RC connector. The planned kit will include two RC connectors, one to attach to the wires from the battery case, and one for the camera USB cable (the servos come ready to connect).

When I built the two SkyShields for Kitty and Scott, I squirted some hot glue around the header for the camera connectors. That should beef them up some. So if you follow the instructions for mounting everything, it should be quite reliable.

Fastie, 2015-06-08:
On further consideration, a ready-made cable would be really nice for kit builders. That is, making your own USB to RC connector cable is not trivial. I just made the two for Scott's and Kitty's rigs. I'm getting good at it, but it's a lot to ask a kit builder to do. 

So a female USB connector on the SkyShield would please a lot of people. I can't find any cables for sale that are ideal. The camera connection should be a right angle mini B (straight works for most cameras, but right angle is the best solution). The SkyShield connection should be micro B, but that cable does not exist. However, there are right angle mini B to straight mini B for sale: <http://www.ebay.com/itm/190925729450>. So if a mini B female connector can be mounted vertically on the SkyShield, it might be worth adding a couple of extra dollars to the BOM. This might work: <http://www.digikey.com/product-detail/en/1734753-1/A107790-ND/4021671>.  
[image]  

There has to be only one of these. The second camera connection can be an RC header. Most people won't use it at all, and those that do can deal with it. It should be vertical because all of the other connectors are vertical and the whole thing becomes a spider explosion if things also have to be connected sideways. It has to be away from the edges of the SkyShield so it can be reached through the gap under the pan servo. The current location of CM1 and CM2 is good, but the fatter USB connector probably requires it to be farther toward the middle of the PCB. It will need some space around it for the fat connector.

Fastie, 2015-06-09:
A simpler way to solve this problem is to construct the custom camera cables differently. I have been attaching RC connectors to the cut end of a USB cable because that makes a very tidy cable. Instead I could use a pre-made RC lead (thin wires with RC connectors at both ends) and cut one connector off. Then the cut ends of the servo lead and the USB cable get spliced together. That is much easier soldering. It also avoids the problem of a stiff USB cable yanking on the connector at the SkyShield. The thin wires of the RC lead are more flexible and won't transfer the motion of the tilting camera as well. The cable could be secured (e.g., with Velcro as per the current design) to make it even more secure.

That strategy requires more common materials (servo leads are cheap and right angle mini B USB cables are a commodity item). So it doesn't add more than 50¢ to the BOM. No change in the SkyShield configuration is required.

Fastie, 2015-06-13:
The camera cable I made for the sixth Saturn V Rig used a right angle USB cable and a servo wire. I chopped them both and spliced them together. It's a much easier build process and it worked great. The flexible servo wires have enough slack that the SkyShield can be pulled out without bothering the Velcro holding the cable.  
[image][image]  
.  
When I had to select a mode by pressing the black button, I did not move the SkyShield, but used a wooden dowel (spare rig leg) that was in my pocket for that purpose. A non-conductive material is recommended.