p-v-o-s/sky-sweep #11: Version 2.0

Fastie, 2015-06-21:
Six SkySweep PCBs arrived in the mail today from OSH Park, simultaneously with DIP switches from Digi-Key. I had to plant the final 50 vegetable starts this afternoon, but after dinner I started soldering.  

[image]
*Doesn't look very pretty without the colorful LEDs, and I tried to make the ground header pins black with a sharpie, and strengthen the camera headers with hot glue. But it works. Mode 7 is selected.*  

Coding the DIP switch is much easier than the tactile button and LEDs -- seven lines of code instead of 40. Everything worked the first time the new sketch was loaded.  

Question: I assume there is no reverse polarity protection on the SkyShield. Should there be? (Yes). In the previous version, the tall capacitor prevented the battery connector from being plugged in wrong. In this version it is quite easy to connect the battery backwards. Is that easy to fix with a diode somewhere?

A warm, gentle rain is caressing my tiny Brussels sprouts and leeks.

dwblair, 2015-06-22:
NICE!

Should we put some LEDs back on the board as optional indicators?

Reverse polarity does indeed sound like a good idea.  Does seem like we can
do this with a simple diode ... will look into it ...

Fastie, 2015-06-22:
LEDs are nice, but my fascination with blinking lights will wear off soon.

A diode will lower the current some, so I guess there is a cost. We are starting with 6V (four alkaline AAA), and the servos need at least 4.8V and would rather have 6V. But if people can fry the Nano by plugging the battery connector in wrong, we should have a diode.

bgamari, 2015-06-22:
@Fastie strictly speaking it will drop the voltage a bit. What you want is a [Schottky diode](https://en.wikipedia.org/wiki/Schottky_diode), which has a very low forward voltage drop.

Fastie, 2015-06-22:
Hooray for Schottky. Something like this?
<http://www.digikey.com/product-detail/en/SR203-TP/SR203-TPCT-ND/950551>
Thanks Ben.

bgamari, 2015-06-22:
Fastie <notifications@github.com> writes:

> Hooray for Schottky. Something like this?
> <http://www.digikey.com/product-detail/en/SR203-TP/SR203-TPCT-ND/950551>
> Thanks Ben.
>
Yep.