p-v-o-s/sky-sweep #12: Version 2.3

Fastie, 2015-06-21:
Using a Sharpie to make the headers for ground pins black is messy. Check out how this is done on the Funduino shield:  
[image]  

I have black, red, and yellow male headers. If the headers for the three servo connections on the SkyShield could be in a 3x3 block instead of a line of nine pins, we could do this. I think this helps the user know how to plug in the connectors and avoids bad mistakes.

We could also put the two camera headers in the same block. The camera connectors need only two pins (G and V), but a third unused pin will make the connection stronger. With the new DIP switch and brilliant use of the Nano's built in pull-up resistor, I think there might be room now for a 3x5 block of headers. 

Can we have this for version 2.3, please?

Fastie, 2015-06-21:
#### Some cosmetic changes for V 2.3 of the PCB:
Let's go with name "SkyShield" instead of "SKY-SWEEP"
The DIP switch seems to be labeled with "SW" and "SW2" which overlap.
We can eliminate the wiki URL on the underside.
Maybe this URL is better for the underside: https://github.com/p-v-o-s/sky-sweep

dwblair, 2015-06-22:
Cool!

Okay, we should put together a list of items for version 2.3 ...

- Cosmetic stuff (above)
- Servo header changes
- Camera header changes
- Fix the wiring on the DIP switches
- ?

Fastie, 2015-06-22:
Is the DIP wiring is already fixed in V 2.2?

Here is a doodle of the new header block:
[image]

BobStevens, 2015-06-22:
There is a locking pad/hole pattern for 0.1" headers in the Sparkfun eagle library. This might ease the assembly and soldering of those headers. The pattern of the pad/holes are slightly staggered and grip the headers in place before soldering.